MicrosoftDocs/windows-driver-docs #2023: dskheap doesn't work on Server 2016 and later

Kerberrrus, 2020-02-04:
This command doesn't work on Server 2016 & 2019. I've posted some details here - https://social.msdn.microsoft.com/Forums/en-US/557ef92f-cb54-4a4b-a34f-18e896d5bb13/failed-getting-grpwinstalist-check-your-win32kbase-symbols-on-windows-10-or-win32-symbols-on?forum=windbg 

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bd70099c-7eae-b6f6-16e5-03b60a62ef46
* Version Independent ID: 7ca031ac-a644-7b5d-5519-439d059f77c0
* Content: [dskheap - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-dskheap#feedback)
* Content Source: [windows-driver-docs-pr/debugger/-dskheap.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-dskheap.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

ralish, 2020-02-29:
Confirming the same issue on Windows 10 v1909. It'd be much appreciated if WinDbg could be updated with support for this command on recent Windows releases. It's probably not a frequently used command, but it's absolutely invaluable for certain debugging scenarios!

DOMARS, 2020-07-25:
@ralish and @Kerberrrus - I too am seeing issues in my testing of !dskheap with recent builds of Windows. The best way to provide product feedback to the debugger team when things are not working the way that they should, is to use this link to file product feedback using the Windows Feedback Hub: 

http://aka.ms/dexex 

The Feedback Hub allows customers to file and up vote product issues. The link automatically sets the product category to Debugging Tools for Windows.

Thanks for reaching out to the Windows Driver Developer Documentation team.

closed 2020-07-25 by DOMARS

ralish, 2020-07-25:
Have opened new issue:
https://github.com/microsoftfeedback/WinDbg-Feedback/issues/54

DOMARS, 2020-07-27:
Great, thanks @ralish