MicrosoftDocs/windows-driver-docs #2026: This page

InventorXtreme, 2020-02-06:
Not detailed enough, no info on what it is, only about how it is a rare error.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4aeee66f-b3f0-5144-87ef-0c0890f2165b
* Version Independent ID: 3119e53b-56bb-5e62-9173-6eb7752f39e9
* Content: [Bug Check 0x10 SPIN_LOCK_NOT_OWNED - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x10--spin-lock-not-owned#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x10--spin-lock-not-owned.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x10--spin-lock-not-owned.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-27:
@InventorXtreme -  Thanks for the input that we should add additional information on Bug Check 0x10. We use customer input like this as well as data on how often the bug checks occur to determine how we allocate our limited documentation team resources.

DOMARS, 2020-12-22:
@InventorXtreme - Thanks for this input on Bug Check 0x10 for orphaned spin locks. I have created and internal work item to track this request. We work to balance the limited resources that we have for debugger docs and the stop code docs in such a way to do the most good, for the largest number of users. For general tips that apply to most stop codes, including this one, see [Blue Screen Data](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/blue-screen-data). I hope you were able to resolve the issue that you ran into and thanks for your interest in the stop code documentation.

closed 2020-12-22 by DOMARS