MicrosoftDocs/windows-driver-docs #2031: 0x8887 missing

MrPepka, 2020-02-08:
I came across the blue screen 0x1A where the first parameter is 0x8887. Can you add the meaning of this parameter on this page?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management?redirectedfrom=MSDN)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-26:
@MrPepka -  Thanks for the input that we should add additional information on Bug Check 0x1A. We use customer input like this as well as data on how often the bug checks occur to determine how we allocate our documentation team resources.

jasonepperly, 2020-03-30:
Added!

DOMARS, 2020-12-18:
@MrPepka, the 0x8887 has been added. Thanks for your interest in improving the bug check documentation.

closed 2020-12-18 by DOMARS