MicrosoftDocs/windows-driver-docs #2038: INF Version section link is wrong

baget, 2020-02-12:
INF Version (in Remarks and also in See Also) section link is wrong, it point to INF AddService and not INF Version
it should point to
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-version-section

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cdb22e15-68dc-98c7-6ef9-eacc7657ad16
* Version Independent ID: 5d242ff3-a365-9ff4-12d7-4cf27a23c099
* Content: [DEVPKEY_Device_ClassGuid - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/devpkey-device-classguid#feedback)
* Content Source: [windows-driver-docs-pr/install/devpkey-device-classguid.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/devpkey-device-classguid.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2020-02-15:
@illfated . please review this PR #2041

tedhudek, 2020-02-19:
Fixed in #2041

closed 2020-02-19 by tedhudek