MicrosoftDocs/windows-driver-docs #2040: it's reliability and security ==> its

GabeSJones, 2020-02-14:
Apostrophe problems in a couple of places

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6298d0a1-1f6b-6e00-5445-dfec501a4430
* Version Independent ID: 42f902a4-b1df-b863-7486-de26c54a205f
* Content: [Driver security checklist - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist?platform=hootsuite)
* Content Source: [windows-driver-docs-pr/driversecurity/driver-security-checklist.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/driversecurity/driver-security-checklist.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2020-02-14:
Thanks, @GabeSJones
It's amazing how it's easy to miss its rightful use.
Fixed in source material. Should be live later today.
Eliot

closed 2020-02-14 by EliotSeattle

DOMARS, 2020-02-14:
Thanks @EliotSeattle!