MicrosoftDocs/windows-driver-docs #2043: 0x109 missing

MrPepka, 2020-02-15:
This time I came across the blue screen 0x109 where the 4th parameter is 0x109. You can add a description of this parameter what does it mean?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7ee6b112-77c3-8aba-d0d3-2c0949682496
* Version Independent ID: c8b2f279-2ea3-bff5-8118-89c1d5b6dd21
* Content: [Bug Check 0x109 CRITICAL_STRUCTURE_CORRUPTION - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x109---critical-structure-corruption?redirectedfrom=MSDN)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x109---critical-structure-corruption.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x109---critical-structure-corruption.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-26:
@MrPepka - Thanks for the input that we should add additional information on Bug Check 0x109. We use customer input like this as well as data on how often the bug checks occur to determine how we allocate our documentation team resources.

SirBorkington337, 2020-05-12:
I have an older system that experienced the same parameter crash last night. A description of this crash would be great.

Tjossul, 2020-05-22:
Same here. I noticed that the first time some days ago.

DOMARS, 2020-12-19:
@MrPepka and @Tjossul - Thanks for the input that on the additional parameter for the 0x109 bug check topic. I have created and internal work item to track the work for this, so will close this issue. As we have limited resources, we use data like this to help guide our work. Good luck with your efforts in blue screen troubleshooting, and thanks for your interest in the bug check documentation.

closed 2020-12-19 by DOMARS