MicrosoftDocs/windows-driver-docs #2044: Provisioning target documentation is unclear

DepthDeluxe, 2020-02-17:
It was very hard to follow the first step of "Provision the target computer".   The instructions said to open the "Driver" menu but did not specify where to find this menu in the UI.  I ultimately found the "configure target devices" shortcut icon just to the right of the debug launch button.

Perhaps including a screenshot here would be helpful.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ce050ea3-d61e-830c-3038-f725eb4c29e1
* Version Independent ID: eb7fc65e-bcfe-dc86-3f8c-336d20eb0c86
* Content: [Provision a computer for driver deployment and testing (WDK 10) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/provision-a-target-computer-wdk-8-1#feedback)
* Content Source: [windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

KeithDobbelaere, 2020-03-29:
I agree.  Thank you for this comment, because I was completely dumbfounded as to what a "Driver" menu was.

cubisttriangle, 2020-05-08:
Same. In VS19, it's in Extensions -> Driver -> Test -> Configure Devices...

tedhudek, 2020-10-23:
This now fixed.  Thanks for the feedback!

closed 2020-10-23 by tedhudek