MicrosoftDocs/windows-driver-docs #2045: Examples would be really helpful.

pankajbharti, 2020-02-20:
Several Articles on JS provide example with each section. 

I am finding it difficult to play around types. Few things i tried: 

I updated the following to pring teh targetType of object a. 

function addTwoValues(a, b)
 {
     host.diagnostics.debugLog("  ", a.targetType, "\n");
     return a + b;
 }

I tried use use: 

dx -v Debugger.State.Scripts.dmf_ScheduledTask.Contents.host.currentProcess.KernelObject.targetType

I tried : 
dx Debugger.State.Scripts.dmf_ScheduledTask.Contents.host.getModuleType(@$foo.ScheduledTaskConfig)

But nothing really worked. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4681f729-986b-1860-b136-f534dfe77dcf
* Version Independent ID: 47f966be-4d2b-70c1-c39f-23d3264b9e09
* Content: [Native Debugger Objects in JavaScript Extensions - Type Objects - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/native-objects-in-javascript-extensions-type-objects#feedback)
* Content Source: [windows-driver-docs-pr/debugger/native-objects-in-javascript-extensions-type-objects.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/native-objects-in-javascript-extensions-type-objects.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-27:
@pankajbharti - Thanks for this input, I will look at the sample code provided here.

In addition to the samples here, I would look at the samples on git hub, for example this one:

https://github.com/microsoft/WinDbg-Samples/blob/master/CodeFlow/CodeFlow.js

DOMARS, 2021-02-04:
@pankajbharti - Thanks for this docs input on the Debug Model JavaScript extensions. I added a code snip that shows the use of APIs that you mentioned in issue #2046. I also have worked to better highlight the new JavaScript code samples now available in https://github.com/microsoft/WinDbg-Samples/. Thanks for this input and good luck with your Debug Model JavaScript development.

closed 2021-02-04 by DOMARS