MicrosoftDocs/windows-driver-docs #2050: Install the App clarification

yves-dolce, 2020-02-22:
Does the current text also applies to application being developed (i.e. when in developer mode). Launching and implicitly deploying from Visual Studio (e.g. F5) won't work? 

Could you please clarify?

Thank you.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6d835c91-424a-a096-ea00-f1dd7c0bc0c4
* Version Independent ID: 1e73f683-6a1e-6794-8831-b6e6b170fa88
* Content: [Hardware Support App (HSA) Steps for App Developers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devapps/hardware-support-app--hsa--steps-for-app-developers#feedback)
* Content Source: [windows-driver-docs-pr/devapps/hardware-support-app--hsa--steps-for-app-developers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devapps/hardware-support-app--hsa--steps-for-app-developers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

tedhudek, 2021-01-21:
Hi again Yves - again, sorry for the delay.  I don't have a lot of context with HSA apps, and I'm not entirely clear on your question.  If  it's still a problem after all this time, can you please try to rephrase the question?

tedhudek, 2021-01-26:
Closing out due to inactivity - please reopen as needed

closed 2021-01-26 by tedhudek