MicrosoftDocs/windows-driver-docs #2053: Error description not possible

Schlorghan, 2020-02-23:
This states that THREAD_STUCK_IN_DEVICE_DRIVER is a problem with a spinning disc. My computer has an SSD and therefore no spinning disc. It only happens when playing YouTube for half an hour so this error is not critical, although it is annoyingly common to ruin the user experience. Thanks for improving the product.

I am sorry to not provide a lot of information to help you fix the problem, I do not know what to provide and don't want to leak any information that might be private, Thanks for  understand.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: fd762fe7-0617-f632-36d0-4eec595f66ff
* Version Independent ID: 72c8c9ec-1645-a307-ae17-c4977b66f57c
* Content: [Bug Check 0xEA THREAD_STUCK_IN_DEVICE_DRIVER - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xea--thread-stuck-in-device-driver#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xea--thread-stuck-in-device-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xea--thread-stuck-in-device-driver.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-25:
@Schlorghan - For assistance with your issue I would suggest you check out one of the support forums. For example this post is describing the same error:

https://answers.microsoft.com/en-us/windows/forum/windows_10-hardware/thread-stuck-in-device-driver-error/d49c3a5d-c6bb-446c-bf17-38d6108711b1

In general, checking for updated drivers is a good idea. The forums can provide more guidance on all of that. Here is a blue screen troubleshooting page that provides general information:

https://support.microsoft.com/en-us/help/14238/windows-10-troubleshoot-blue-screen-errors 

These docs are for the developers that are working to fix the code that caused the issue and have access to the related source code. Good luck in fixing the issue with your PC.

closed 2020-02-25 by DOMARS

Schlorghan, 2020-02-25:
Thank you for this great Customer-service, Letting you know that I
appreciate your response.

Thanks!