MicrosoftDocs/windows-driver-docs #2055: CRITICAL_STRUCTURE_CORRUPTION (0x109) missing type 0x114

chupAkabRRa, 2020-02-24:
While using DDIMon tool (https://github.com/tandasat/DdiMon) I faced a BSOD 0x109 (CRITICAL_STRUCTURE_CORRUPTION) with a type of corrupted region equals to 0x114. Can you add a description of this parameter and what does it mean, please?

DOMARS, 2020-02-27:
@chupAkabRRa - If possible can you open this issue on that stop code topic? 

https://docs.microsoft.com/windows-hardware/drivers/debugger/bug-check-0x109---critical-structure-corruption

That will provide better visibility.

DOMARS, 2020-12-23:
@chupAkabRRa - Thanks for the input that we should add additional information on Bug Check 0x109 on parameter. We use customer input like this as well as data on how often the bug checks occur to determine how we allocate our documentation team resources. As you may know we provide general developer guidance in topics such as https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/blue-screen-data. In general, if a specific driver continues to cause this stop code, the developer of that driver will need to update their code. If you don’t have the source code, you can try updating the driver, if a newer version is available. We will use your input and other data to determine if we can provide additional information on this stop code parameter that is broadly applicable. Again, thanks for the input, and good luck with your issue.

closed 2020-12-23 by DOMARS