MicrosoftDocs/windows-driver-docs #2059: Command with wrong text indication

arsu-leo, 2020-02-26:
Section 5: Note "Understand source code symbol requirements", point 3
The command in this paragraph as a counter slash "\", running the command with it returns no output.
The example just below has no "\", running it like that gives the expected output
"[...]type x ECHO!Echo\*" is wrong. The "\" must not be there.
The example is correct:
0: kd&gt; x ECHO!Echo*

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f5abd01e-3c7e-11a8-a6c6-62bfba0ef8b6
* Version Independent ID: dceb81b3-b20f-91d1-85d0-fa68c762482b
* Content: [Debug Universal Drivers - Step-by-Step Lab (Echo Kernel Mode) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-)
* Content Source: [windows-driver-docs-pr/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-27:
@arsu-leo - Thanks for pointing out this typo with the escape character being used with the asterisk, it is now fixed. Good luck in your debugging adventures.

closed 2020-02-27 by DOMARS

DOMARS, 2020-02-28:
@arsu-leo - I fixed a few other places yesterday as well. But please do let me know if you see any that I missed. Thanks for your help in improving this lab.