MicrosoftDocs/windows-driver-docs #2065: EV certificate required before Signing the package in HLK Studio or after submitting the package at Windows Dev Center Dashboard?

IqraJaved, 2020-03-02:
I have developed a driver for windows that works similar to OpenVPN. I wanted to get it signed by the microsoft so I have tested it using VHLK 1809. 

Some of the tests are failed (that I think are not related to the driver). The failed test cases are: 
1- DF - Concurrent Hardware And Operating System (CHAOS) Test (Reliability) - System Crashed
2- NDISTest 6.0 - [1 Machine] - 1c_FaultHandling (Passed: 21, Failed: 2)
3- NDISTest 6.0 - [1 Machine] - 1c_IOCTLCoverage (Passed: 168, Failed: 4)
4- NDISTest 6.0 - [1 Machine] - 1c_Mini6PerfSend (Passed: 75, Failed: 73)
5- NDISTest 6.0 - [1 Machine] - 1c_Mini6Send (Passed: 76, Failed: 73)
6- NDISTest 6.0 - [1 Machine] - 1c_NdisRequestCov (Passed: 11, Failed: 3)
7- NDISTest 6.0 - [1 Machine] - 1c_OidsDeviceControl (Passed: 120, Failed: 4)
8- Static Tools Logo Test (Passed: 4, Failed: 1)

I have made a submission package by selecting the "Do Not Sign" Option, We have applied for the EV certificate and hoping to get it soon. 

My question is:  
Should I submit this package (Made with **Do Not Sign** option ) along with an EV certificate to Microsoft Partner Center for Driver Signing)?
Or Should I make the submission package by providing the certificate file and using **Use a certificate file** option in the HLK studio ? 

Kindly help me regarding this confusion as we are almost in final stages of driver signing. 
Regards.

IqraJaved, 2020-03-06:
Is anyone looking into this issue?

EliotSeattle, 2020-03-06:
Hi @IqraJaved 

The Hardware Dev Center Support folks are the right people to help investigate an issue like this.
1. **First**, sign into your hardware dev center account, and then 
2. Open a Hardware Dev Center support case for investigation. 
3. Go to 
[https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq)
and look for: 
 
**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). Select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

I hope this helps.
Eliot

IqraJaved, 2020-03-09:
Thank you @EliotSeattle for the information.