MicrosoftDocs/windows-driver-docs #2067: Error 1297

GabeSJones, 2020-03-03:
This doesn't appear to happen in the 1903 WDK infverif, but the infverif in use by the Windows Hardware Dev Dashboard appears to be a more recent version.  (Sure would be nice to get my hands on that!)  It is throwing the following for one of our drivers:

Error 1297 in niapdummy.inf, line 0 : Device driver does not install on any devices, use primitive driver if this is intended.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 48275f87-4086-1ca5-a190-c741f29ad348
* Version Independent ID: 5481885d-3063-527b-a85e-6a898dbcb990
* Content: [INF Validation Errors and Warnings - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/inf-validation-errors-and-warnings#syntax-in-the-inf-file-1100-1299)
* Content Source: [windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

GabeSJones, 2020-03-03:
OK, I tried the infverif from a preview WDK.  I got the 1927 error.  I also got the following warning:

WARNING(1324): [Version] section should specify PnpLockdown=1.

zlockard, 2020-03-03:
@GabeSJones Can you share the driver?  This error basically means that you've got a driver that doesn't do anything

GabeSJones, 2020-03-03:
> @GabeSJones Can you share the driver? This error basically means that you've got a driver that doesn't do anything

I can't share the driver, but, yes, it doesn't do anything.  We use it for some internal build smoke testing.  Uploading it to MS was unintentional, but when I got the previously-unfamiliar-to-me error & warning I wanted to make sure the docs were updated.

zlockard, 2020-03-03:
Understood.  We'll update the docs.  If you want to have a driver that just performs some operations and doesn't install on any device, you should use [DefaultInstall.NT(ARCH)] instead of [Manufacturer].

GabeSJones, 2020-03-03:
Perfect!  Thank you for the info and for your time and attention.

KylePhoria, 2020-03-19:
Just run into the same issue with 20H1 WDK. InfVerif failed for Driver. Partial Log:Error 1297 in xxx.inf. line 0 : Device driver does not install on any devices, use primitive driver if this is intended.
It seems MS has revised the policy of InfVerif...

TPCaZan, 2020-03-19:
Same problem here.
It would be nice to get any advice how to circumvent this error. Are we supposed to convert our driver into 'primitive drivers' now?
If so, how can we make our driver backward compatible with Windows 7 & 8.1? Because they don't support primitive drivers.

zlockard, 2020-03-19:
You're supposed to convert it to a primitive driver if it's not a device driver, yes.

Primitive drivers are supported down to Vista.  This doc covers legacy compatibility at the bottom:  https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/creating-a-primitive-driver

TPCaZan, 2020-03-20:
Im my case, the affected driver is a virtual printer driver. And I already tried to make a primitive one, following the guidelines in the article you mentioned.
I also followed the recommendations for legacy compatibility.
The new resulting Inf-file passed the InfVerifier tests and installed properly on Windows 10.
But it fails to install on all OSs older than that. For me, it looks like the older OSs rely on the [Manufacturer] section, which is not allowed for primitive drivers.

Sorry for being inquisitive, but are you sure that priitive drivers are supported on older OSs? Does it require a specific update of some kind?

zlockard, 2020-03-20:
Ah, sorry for the lack of clarity.  The Windows 10 feature for primitive drivers was to converge the two install paths for (previously unnamed) primitive drivers and device drivers.  Pre-Win10, these two types of drivers had different install mechanisms.  Pre-Win10, you would use this install pattern for primitive drivers:
setupapi!SetupCopyOEMInf    <- To put the INF in the driver store
setupapi!InstallHInfSection   <- Specify DefaultInstall as the section

This same pattern works in Win10, or you can use the device driver install APIs in Win10.

draftpin, 2020-04-06:
After last update we can't sign my driver for more than one month!
My structure is simple:
x64/acdrv.inf
x64/acdrv.sys

First problem was solved by changing
[Manufacturer]
to
[DefaultInstall.NT(ARCH)]

But after that i'm getting another error at "Catalog creation"
"No installation INF found in the root path of the driver. For the driver to be digitally signed, and for it to install properly the installation inf must be in the root of each driver path you have provided. If you have separate folders with different driver packages for different languages, operating systems or device categories then specify each driver set as a separate driver. Each driver package must be selected separately and must contain an installation INF."

This error is also reproduced by inf2cat utility

If i change back
[DefaultInstall.NT(ARCH)] to [Manufacturer] and check with inf2cat utility, then inf2cat utility test passes but i'm getting first error "Device driver does not install on any devices, use primitive driver if this is intended.". It seems, that this error is produced by new infverif tool that is not possible to download and check inf file manually. Because my last ewdk inverif tool does not give any errors on inf.

Support man (his name is Jason) is apologizing every day without any real help for 1 month with last error, i've also tried everything and still no result to pass all the tests.
How to fix? 

I also generated and included cat file into CAB using makecat tool, but it changes nothing.

How is it checked whether my driver is primitive or not? What should be added to make system thinking that it's not primitive, because in this case we are able to pass catalog creation (inf2cat) test and finish the job.

Or may be someone is already signing and can share working example of INF?

zlockard, 2020-04-06:
Can you share the inf2cat command you're running?  I'm not able to reproduce this issue

draftpin, 2020-04-06:
I've uploaded full package here:
http://gamedrive.pro/upload/uploads/testInf.rar
Password: INh5qbsJcl0y28d8W5NDd6UmcKn45a
Run test\inf_verify.cmd and you will see everything

zlockard, 2020-04-07:
Ah I see the confusion.  I will write this up more extensively.  [DefaultInstall] is both an entry point **and** a DDInstall section.

Here's how you make the transition:
```
[Manufacturer]
%OnMoon%=GameGuard, NTx86, NTamd64

[GameGuard.NTx86]
acdrv = ACDRV_Install

[GameGuard.NTamd64]
acdrv = ACDRV_Install

[ACDRV_Install]
OptionDesc = %ServiceDesc%
Copyfiles = ACDRV_Files

[ACDRV_Install.Services]
AddService = acdrv,,ACDRV_Service

```
Should become:
```
[DefaultInstall.NTamd64]
OptionDesc = %ServiceDesc%
Copyfiles = ACDRV_Files

[DefaultInstall.NTamd64.Services]
AddService = acdrv,,ACDRV_Service

[DefaultInstall.NTx86]
OptionDesc = %ServiceDesc%
Copyfiles = ACDRV_Files

[DefaultInstall.NTx86.Services]
AddService = acdrv,,ACDRV_Service
```

draftpin, 2020-04-07:
Thank you so much! It WORKS!!! 
I don't understand why those guys from official support can't solve this for a month

tedhudek, 2020-04-07:
Docs updated in #2153.  Thanks all!

closed 2020-04-07 by tedhudek

venkataramana-motupally, 2020-04-16:
Hi .. i too got smae kind of error .. i was able to WHQL  sign the xps drivers .. but unable to install driver x64 or x86 based machines...
[image]

Any help will be appreciated .

zlockard, 2020-04-16:
Primitive drivers are for drivers that don't install on any devices. If it's for a device like a printer, it shouldn't be a primitive driver

venkataramana-motupally, 2020-04-17:
hi zlockard,  

thank you for the info.
When I try sign xpsdrv sample it's giving me 1297 error in whql signing.  but same inf installing with test sign or win7 and all.   Can u Pls guide on this... 

Thanks, 
Venkat

danthecowboy, 2020-05-11:
I've also an issue with device drivers and the error 1297 from the infver tool (you get the error only if you use the insider version for 20H1). Since I'm getting this error I can't get any signing for my printer driver. Do you know an example of a working universal device driver? I can't use a primitive driver for that.

zlockard, 2020-05-11:
The 1297 error only occurs when your INF doesn't install on any devices.  You have to add some hardware IDs for your INF to install on.  If you have a scenario where this isn't the case, I'd be interested in understanding it

danthecowboy, 2020-05-12:
I didn't have a hardware ID for my INF since it is also a virtual printer driver. My INF should be as compatible as possible with old windows version so the manufactor section looks like this:

```
[Manufacturer]
"SEAL Systems"=Standard,NTx86,NTamd64,NTamd64.6.0,NTx86.6.0
;
; Model sections. 

[Standard]
"SEAL Systems PS OMS Generic"                    = NT5_SEALPSOMS_GENERIC,,
"SEAL Systems PS OMS Office"                     = NT5_SEALPSOMS_OFFICE,,
"SEAL Systems PS OMS CAD"                        = NT5_SEALPSOMS_CAD,,
"SEAL Systems PS OMS Convert"                    = NT5_SEALPSOMS_CONVERT,,

[Standard.NTamd64]
"SEAL Systems PS OMS Generic"                    = NT5X64_SEALPSOMSGENERIC,,
"SEAL Systems PS OMS Office"                     = NT5X64_SEALPSOMSOFFICE,,
"SEAL Systems PS OMS CAD"                        = NT5X64_SEALPSOMSCAD,,
"SEAL Systems PS OMS Convert"                    = NT5X64_SEALPSOMSCONVERT,,

[Standard.NTx86]
"SEAL Systems PS OMS Generic"                    = NT5_SEALPSOMS_GENERIC,,
"SEAL Systems PS OMS Office"                     = NT5_SEALPSOMS_OFFICE,,
"SEAL Systems PS OMS CAD"                        = NT5_SEALPSOMS_CAD,,
"SEAL Systems PS OMS Convert"                    = NT5_SEALPSOMS_CONVERT,,

[Standard.NTamd64.6.0]
"SEAL Systems PS OMS Generic"                    = NT6X64_SEALPSOMSGENERIC,,
"SEAL Systems PS OMS Office"                     = NT6X64_SEALPSOMSOFFICE,,
"SEAL Systems PS OMS CAD"                        = NT6X64_SEALPSOMSCAD,,
"SEAL Systems PS OMS Convert"                    = NT6X64_SEALPSOMSCONVERT,,

[Standard.NTx86.6.0]
"SEAL Systems PS OMS Generic"                    = NT6_SEALPSOMS_GENERIC,,
"SEAL Systems PS OMS Office"                     = NT6_SEALPSOMS_OFFICE,,
"SEAL Systems PS OMS CAD"                        = NT6_SEALPSOMS_CAD,,
"SEAL Systems PS OMS Convert"                    = NT6_SEALPSOMS_CONVERT,,
```
It is mostly inspired by the WDK examples from here:
https://github.com/microsoft/Windows-driver-samples/blob/master/print/OEM%20Printer%20Customization%20Plug-in%20Samples/C%2B%2B/oemdll.inf

Also this INF passes the HCK Tests for Driver Submission, but can't get a signature because of the 1297 error from infver. Thanks for having a look ;-)

zlockard, 2020-05-26:
Sorry for the delay, I had to sync with the printer folks for guidance here.

For a virtual printer driver, the guidance is that you must specify a unique string for the hardware ID on these devices, an obvious choice being to generate a unique GUID.  For your particular case, you could generate 4 GUIDs, one for each of the devices, and reuse them for each specified OS version.

`
[Standard]
"SEAL Systems PS OMS Generic"                    = NT5_SEALPSOMS_GENERIC,{GUID1}
"SEAL Systems PS OMS Office"                     = NT5_SEALPSOMS_OFFICE,{GUID2}
"SEAL Systems PS OMS CAD"                        = NT5_SEALPSOMS_CAD,{GUID3}
"SEAL Systems PS OMS Convert"                    = NT5_SEALPSOMS_CONVERT,{GUID4}

[Standard.NTamd64]
"SEAL Systems PS OMS Generic"                    = NT5X64_SEALPSOMSGENERIC,{GUID1}
"SEAL Systems PS OMS Office"                     = NT5X64_SEALPSOMSOFFICE,{GUID2}
"SEAL Systems PS OMS CAD"                        = NT5X64_SEALPSOMSCAD,{GUID3}
"SEAL Systems PS OMS Convert"                    = NT5X64_SEALPSOMSCONVERT,{GUID4}

<etc>
`

danthecowboy, 2020-06-03:
Thanks zlockard, I was in touch with the HCK support team. You need the GUID for the infverif tool that's right, but you also need an additional comma for passing the hck inf checks. So this is the working solution for me (passing hck tests and submission):

```
[Standard]

"SEAL Systems PS OMS Generic"                    = NT5_SEALPSOMS_GENERIC,,{GUID1}
"SEAL Systems PS OMS Office"                     = NT5_SEALPSOMS_OFFICE,,{GUID2}
"SEAL Systems PS OMS CAD"                        = NT5_SEALPSOMS_CAD,,{GUID3}
"SEAL Systems PS OMS Convert"                    = NT5_SEALPSOMS_CONVERT,,{GUID4}

[Standard.NTamd64]
"SEAL Systems PS OMS Generic"                    = NT5X64_SEALPSOMSGENERIC,,{GUID1}
"SEAL Systems PS OMS Office"                     = NT5X64_SEALPSOMSOFFICE,,{GUID2}
"SEAL Systems PS OMS CAD"                        = NT5X64_SEALPSOMSCAD,,{GUID3}
"SEAL Systems PS OMS Convert"                    = NT5X64_SEALPSOMSCONVERT,,{GUID4}
```

tedhudek, 2020-06-03:
Thanks for circling back, @danthecowboy.  If there is anything specific that you think should be added to the docs, please consider opening a PR with the change.

zlockard, 2020-06-03:
Just to be clear, you're actually generating 4 guids here and not using those placeholders :)

hypervisor, 2020-06-05:
Have the same issue, I create a new WDM project with nothing special, default settings and it doesn't compile because "Device driver does not install on any devices, use primitive driver if this is intended". I recently upgraded my Visual Studio and WDK installations, didn't have this issue before. I could just build it after creating a test signing certificate. I can't do that now. Any suggestions?

zlockard, 2020-06-05:
You need to make the driver install on some set of devices.  You could add dummy hardware IDs to your INF to start with if you'd like

hypervisor, 2020-06-05:
Sorry if this is a stupid question, but how would I do that? And why did I not have to do that before?

zlockard, 2020-06-05:
If you're making a device driver:  In the INF or INX file in the project with the error, you should add some hardware IDs.  [Here](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-models-section) are the MSDN docs on this.  If you're just trying to get an initial compile going and don't even have a device to install on yet, you can just make up some hardware IDs.  A random string like "DummyHardwareId" will work fine.  If you have a device to install on, you'll want to use a hardware ID specified by the device.

If you're making some kind of driver that isn't intended to install on devices:  you'll want to make a primitive driver.  Guidance for that is [here](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/creating-a-primitive-driver).

These are new checks designed to prevent people from creating drivers that don't work as intended.  Unfortunately you do have an extra step when creating a project, but we were seeing a large number of drivers that failed after upgrade because their INF was not specified correctly

stonedreamforest, 2020-08-10:
I also encountered this problem, It can compile normally on my machine, this error appears in github actions.
```INF
;
; MyDriver1.inf
;

[Version]
Signature="$WINDOWS NT$"
Class=System
ClassGuid={4d36e97d-e325-11ce-bfc1-08002be10318}
Provider=%ManufacturerName%
DriverVer=
CatalogFile=MyDriver1.cat

[DestinationDirs]
DefaultDestDir = 12

[SourceDisksNames]
1 = %DiskName%,,,""

[SourceDisksFiles]

[Manufacturer]
%ManufacturerName%=Standard,NT$ARCH$

[Standard.NT$ARCH$]

[Strings]
ManufacturerName="<Your manufacturer name>" ;TODO: Replace with your manufacturer name
ClassName=""
DiskName="MyDriver1 Source Disk"

```
This default generated file, how can I modify it.

LusKrew, 2020-10-03:
"You need to make the driver install on some set of devices. You could add dummy hardware IDs to your INF to start with if you'd like", WHY? why do i need to add dumy hardware ID to build a simple hello world driver?

LusKrew, 2020-10-03:
```
;
; HelloWorld.inf
;

[Version]
Signature="$WINDOWS NT$"
Class=System
ClassGuid={4d36e97d-e325-11ce-bfc1-08002be10318}
Provider=%ManufacturerName%
DriverVer=
CatalogFile=HelloWorld.cat
PnpLockDown=1

[DestinationDirs]
DefaultDestDir = 12

[SourceDisksNames]
1 = %DiskName%,,,""

[SourceDisksFiles]

[Manufacturer]
%ManufacturerName%=Standard,NT$ARCH$

[Standard.NT$ARCH$]

[Strings]
ManufacturerName="<Your manufacturer name>" ;TODO: Replace with your manufacturer name
ClassName=""
DiskName="HelloWorld Source Disk"

```

HOW am i supposed to change this so i can build a SIMPLE hello world? because the change mentioned above does not work..

zlockard, 2020-10-05:
The INF file defines how the driver is to be installed on the system, the driver needs a target to install onto.  In your INF, you should need to add a [DDInstall](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-ddinstall-section) section and add hardware IDs into your [Models section](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-models-section) (which is [Standard.NT$ARCH$]).  Alternatively you could convert [Manufacturer] into [DefaultInstall.NT$ARCH$] and treat it like a DDInstall section