MicrosoftDocs/windows-driver-docs #2069: OS Values for the 20H1 release?

GabeSJones, 2020-03-04:
What will the values for the 20H1 Windows release be?  I'm guessing WINDOWS_v100[_X64|_ARM]_VB_FULL, but would like confirmation/correction.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1c20769b-d969-a51a-4eb3-f9cc27875f28
* Version Independent ID: 853fe2a9-d654-98a3-7635-c30a450cdbdc
* Content: [Get product data - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-product-data#list-of-operating-system-codes)
* Content Source: [windows-driver-docs-pr/dashboard/get-product-data.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-product-data.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-03-05:
Thanks for the reminder, @GabeSJones 
I'm working on the verification.
Eliot

EliotSeattle, 2020-03-25:
Hi @GabeSJones 
Yes, you are correct. However, you'll have to wait to use the new designation until that kit is released. We'll update the docs before then, too.
Thanks!
Eliot

closed 2020-03-25 by EliotSeattle