MicrosoftDocs/windows-driver-docs #2086: Warning / Error about a missing service section

arsu-leo, 2020-03-13:
Warning	1296	Hardware 'v1Sample0.0.0.1' does not have an associated service using install section 'DriverInstall'
On some Windows version this would turn into an error to the sample can't be used
https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/inf-validation-errors-and-warnings#syntax-in-the-inf-file-1100-1299

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 227ceb38-396a-aa19-6304-be3df9b0ca08
* Version Independent ID: 2f34593f-aa20-384a-b1bb-c8a962ec4706
* Content: [Building a Basic v4 Printer Driver - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/print/building-a-basic-v4-printer-driver)
* Content Source: [windows-driver-docs-pr/print/building-a-basic-v4-printer-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/building-a-basic-v4-printer-driver.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

Cymoki, 2020-04-08:
Thanks for pointing this out, I have filed a bug for this to be fixed.

barrygolden, 2021-01-27:
Closing #2086 as there is an internal work item to address the warning/error.

closed 2021-01-27 by barrygolden