MicrosoftDocs/windows-driver-docs #2087: One possible typo

TCoherence, 2020-03-14:
[In the last but one sentence, it is said that _The miniport driver must set the **MinLookaheadSplitSize** and **MinLookaheadSplitSize** members to zero._. Maybe it should be Max and Min?]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 31113831-a970-2f6f-eac6-68db6944ba1f
* Version Independent ID: 965f6988-bdcf-efdb-4e2e-e36451600e65
* Content: [Security Issues with NDIS Virtual Machine (VM) Shared Memory - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/security-issues-with-ndis-virtual-machine--vm--shared-memory#feedback)
* Content Source: [windows-driver-docs-pr/network/security-issues-with-ndis-virtual-machine--vm--shared-memory.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/security-issues-with-ndis-virtual-machine--vm--shared-memory.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

RAJU2529, 2020-03-20:
@PRMerger18 . @billlattimer . Please review the created PR #2103

EliotSeattle, 2020-03-27:
Reviewed and merged PR #2103.
Max for the win!
Thanks, @TCoherence & @RAJU2529!

closed 2020-03-27 by EliotSeattle