MicrosoftDocs/windows-driver-docs #2096: Wrong hyperlink redirect of "Hotspot Authentication Sample"

mxlius, 2020-03-19:
Hyper link redirect to portal page of docs, correct redirect destination is https://docs.microsoft.com/en-us/samples/microsoft/windows-universal-samples/hotspotauthentication/

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7907a978-75e6-42e5-af65-4b0234260458
* Version Independent ID: 7b9e310b-01e7-48c3-d139-fcdad5f2c74e
* Content: [Review the hotspot authentication sample - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/review-the-hotspot-authentication-sample#feedback)
* Content Source: [windows-driver-docs-pr/mobilebroadband/review-the-hotspot-authentication-sample.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/mobilebroadband/review-the-hotspot-authentication-sample.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

EliotSeattle, 2020-03-19:
Thanks @mxlius for finding this.
@billlattimer the link to **Mobile Broadband Provisioning sample** also goes to the landing page, and I don't see it in the repo. Can you check to see if that was decommissioned or renamed and either fix or remove the link?
Eliot

billlattimer, 2020-03-20:
@EliotSeattle looking; that sample is currently under revision.

RAJU2529, 2020-03-20:
@mxlius . @billlattimer . Just now  i checked  the  hyper link of  **Hotspot Authentication Sample**
is  correct.  I conformed it . Also i checked the hyper link of **Mobile broadband sample** is  direct to the complete  sample portal page.
 So i am going to  create PR for correctlink now

RAJU2529, 2020-03-20:
@EliotSeattle . @billlattimer . just now i created PR #2102 .
so please review is possible 

**https://github.com/MicrosoftDocs/windows-driver-docs/pull/2102**

aviviano, 2020-10-29:
Thank you @mxlius for letting us know about this issue. We've updated the link and will close this issue.

closed 2020-10-29 by aviviano