MicrosoftDocs/windows-driver-docs #2100: Parameter 1 values

LuigiBruno, 2020-03-20:
Which are the possible values for the parameter 1 and their corresponding meanings?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ceedbb73-ac5a-74bd-5747-bff43f2b639b
* Version Independent ID: 4118fb85-66db-5598-0512-fa2315bd30c7
* Content: [Bug Check 0x12B FAULTY_HARDWARE_CORRUPTED_PAGE - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x12b---faulty-hardware-corrupted-page?source=docs#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x12b---faulty-hardware-corrupted-page.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x12b---faulty-hardware-corrupted-page.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-12-23:
@LuigiBruno - Thanks for your input on this stop code. I worked with the dev team on this and as it turns our there are two sets of parameters for this stop code. Which set you use is determined if param 3 and 4 are zero. Good luck with your stop code troubleshooting, and thanks again for this issue.

closed 2020-12-23 by DOMARS

LuigiBruno, 2021-01-18:
> @LuigiBruno - Thanks for your input on this stop code. I worked with the dev team on this and as it turns our there are two sets of parameters for this stop code. Which set you use is determined if [...]

Thank you, @DOMARS.