MicrosoftDocs/windows-driver-docs #2106: Modify for Windows 10 and Visual Studio 2019

ahidaka, 2020-03-21:
The page should be modified to for Windows 10 and Visual Studio 2019. And some of the sample code doesn't work any longer. The sample codes should be replaced as followings with my sample code. My tested code is at my repository as followings.

https://github.com/ahidaka/WinUSB_Collection/tree/master/WinUSBApplication

## The modification point-1.
### Prerequisites
Add Windows 10.

## The modification point-2.
### Step 2: Query the Device for USB Descriptors

Replace 2nd sample code, QueryDeviceEndpoints with this.
```C
struct PIPE_ID
{
    UCHAR  PipeInId;
    UCHAR  PipeOutId;
};

BOOL QueryDeviceEndpoints(WINUSB_INTERFACE_HANDLE hDeviceHandle, PIPE_ID* pipeid)
{
    if (hDeviceHandle == INVALID_HANDLE_VALUE)
    {
        return FALSE;
    }

    BOOL bResult = TRUE;

    USB_INTERFACE_DESCRIPTOR InterfaceDescriptor;
    ZeroMemory(&InterfaceDescriptor, sizeof(USB_INTERFACE_DESCRIPTOR));

    WINUSB_PIPE_INFORMATION  Pipe;
    ZeroMemory(&Pipe, sizeof(WINUSB_PIPE_INFORMATION));

    bResult = WinUsb_QueryInterfaceSettings(hDeviceHandle, 0, &InterfaceDescriptor);

    if (bResult)
    {
        for (int index = 0; index < InterfaceDescriptor.bNumEndpoints; index++)
        {
            bResult = WinUsb_QueryPipe(hDeviceHandle, (UCHAR)0, (UCHAR)index, &Pipe);

            if (bResult)
            {
                if (Pipe.PipeType == UsbdPipeTypeControl)
                {
                    printf("Endpoint index: %d, Pipe type: Control, Pipe ID: %X.\n", index, Pipe.PipeId);
                }
                if (Pipe.PipeType == UsbdPipeTypeIsochronous)
                {
                    printf("Endpoint index: %d, Pipe type: Isochronous, Pipe ID: %X.\n", index, Pipe.PipeId);
                }
                if (Pipe.PipeType == UsbdPipeTypeBulk)
                {
                    if (USB_ENDPOINT_DIRECTION_IN(Pipe.PipeId))
                    {
                        printf("Endpoint index: %d, Pipe type: Bulk IN, Pipe ID: %X.\n", index, Pipe.PipeId);
                        pipeid->PipeInId = Pipe.PipeId;
                    }
                    if (USB_ENDPOINT_DIRECTION_OUT(Pipe.PipeId))
                    {
                        printf("Endpoint index: %d, Pipe type: Bulk OUT, Pipe ID: %X.\n", index, Pipe.PipeId);
                        pipeid->PipeOutId = Pipe.PipeId;
                    }

                }
                if (Pipe.PipeType == UsbdPipeTypeInterrupt)
                {
                    printf("Endpoint index: %d, Pipe type: Interrupt, Pipe ID: %X.\n", index, Pipe.PipeId);
                }
            }
            else
            {
                continue;
            }
        }
    }

    return bResult;
}
```

## The modification point-3.
### Step 6: Implement Main
Replace sample code, _tmain with this.
```C
LONG __cdecl
_tmain(
    LONG     Argc,
    LPTSTR * Argv
    )
/*++

Routine description:

    Sample program that communicates with a USB device using WinUSB

--*/
{
    DEVICE_DATA           deviceData;
    HRESULT               hr;
    USB_DEVICE_DESCRIPTOR deviceDesc;
    BOOL                  bResult;
    BOOL                  noDevice;
    ULONG                 lengthReceived;

    PIPE_ID PipeID;
    UCHAR DeviceSpeed;
    ULONG cbSize = 0;

    UNREFERENCED_PARAMETER(Argc);
    UNREFERENCED_PARAMETER(Argv);

    //
    // Find a device connected to the system that has WinUSB installed using our
    // INF
    //
    hr = OpenDevice(&deviceData, &noDevice);

    if (FAILED(hr)) {

        if (noDevice) {

            wprintf(L"Device not connected or driver not installed\n");

        } else {

            wprintf(L"Failed looking for device, HRESULT 0x%x\n", hr);
        }

        return 0;
    }

    //
    // Get device descriptor
    //
    bResult = WinUsb_GetDescriptor(deviceData.WinusbHandle,
                                   USB_DEVICE_DESCRIPTOR_TYPE,
                                   0,
                                   0,
                                   (PBYTE) &deviceDesc,
                                   sizeof(deviceDesc),
                                   &lengthReceived);

    if (FALSE == bResult || lengthReceived != sizeof(deviceDesc)) {

        wprintf(L"Error among LastError %d or lengthReceived %d\n",
                FALSE == bResult ? GetLastError() : 0,
                lengthReceived);
        CloseDevice(&deviceData);
        return 0;
    }

    bResult = GetUSBDeviceSpeed(deviceData.WinusbHandle, &DeviceSpeed);
    if (!bResult)
    {
        goto done;
    }

    bResult = QueryDeviceEndpoints(deviceData.WinusbHandle, &PipeID);
    if (!bResult)
    {
        goto done;
    }

    bResult = SendDatatoDefaultEndpoint(deviceData.WinusbHandle);
    if (!bResult)
    {
        goto done;
    }

    bResult = WriteToBulkEndpoint(deviceData.WinusbHandle, &PipeID.PipeOutId, &cbSize);
    if (!bResult)
    {
        goto done;
    }

    bResult = ReadFromBulkEndpoint(deviceData.WinusbHandle, &PipeID.PipeInId, cbSize);
    if (!bResult)
    {
        goto done;
    }

    system("PAUSE");

    //
    // Print a few parts of the device descriptor
    //
    wprintf(L"Device found: VID_%04X&PID_%04X; bcdUsb %04X\n",
            deviceDesc.idVendor,
            deviceDesc.idProduct,
            deviceDesc.bcdUSB);
done:
    CloseDevice(&deviceData);
    return 0;
}
```
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ab9a2103-b49a-305b-674d-eeb241944822
* Version Independent ID: 6f7d87a9-126e-6a3f-3087-0005482a5f60
* Content: [How to Access a USB Device by Using WinUSB Functions - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/using-winusb-api-to-communicate-with-a-usb-device)
* Content Source: [windows-driver-docs-pr/usbcon/using-winusb-api-to-communicate-with-a-usb-device.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/using-winusb-api-to-communicate-with-a-usb-device.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

RAJU2529, 2020-05-13:
@billlattimer . Sir, please look into this current issue, because there is no reply here. Thank you

tedhudek, 2021-02-24:
Hi @ahidka, thanks for the feedback and apologies for the delay.  If you'd like to open a pull request with your suggested changes into the .md target file, I can ask someone here to review the code changes.  Thank you!

closed 2021-02-24 by tedhudek