MicrosoftDocs/windows-driver-docs #2110: Little misspell in link | remode instead of remote | https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/remode-debugging-using-windbg

mieshki, 2020-03-23:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4d8aad5b-94a4-f88b-211d-f7675ddc36b1
* Version Independent ID: a0743e12-91d4-217f-d1cc-e7d6e8bb871e
* Content: [Remote Debugging Using WinDbg - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/remode-debugging-using-windbg#feedback)
* Content Source: [windows-driver-docs-pr/debugger/remode-debugging-using-windbg.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/remode-debugging-using-windbg.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

Shipleaves, 2020-04-20:
This issue is fixed here:
https://github.com/MicrosoftDocs/windows-driver-docs/pull/2181

DOMARS, 2020-04-21:
@mieshki - Closing this issue as the topic name has been changed to be remote. Thanks for your interest in the debugger docs.

closed 2020-04-21 by DOMARS