MicrosoftDocs/windows-driver-docs #2112: 0x8886 missing

MrPepka, 2020-03-23:
I came across the blue screen 0x1A where the first parameter is 0x8886. Can you ask for a description of this parameter?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management?redirectedfrom=MSDN)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

jasonepperly, 2020-03-25:
Looking into this

jasonepperly, 2020-03-29:
Added!

DOMARS, 2020-05-01:
@MrPepka - This parameter description is now added, so closing this issue.

closed 2020-05-01 by DOMARS