MicrosoftDocs/windows-driver-docs #2117: download location

shuhei-saito-sii-co-jp, 2020-03-25:
I could not find download location of language monitor(Pjlmon.dll).
Please tell me the download location

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0a798e40-954b-c521-34ef-7802d81149e0
* Version Independent ID: eff99c08-f9d0-eb98-2e5a-10ea7ae82626
* Content: [Language Monitors - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/print/language-monitors)
* Content Source: [windows-driver-docs-pr/print/language-monitors.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/language-monitors.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

RAJU2529, 2020-03-25:
@shuhei-saito-sii-co-jp . Downloaded and install Windows Driver Kit (WDK).
You can find that particular file

barrygolden, 2020-06-17:
Thanks for filing this issue @shuhei-saito-sii-co-jp!

The [Sample Language Monitor](https://docs.microsoft.com/windows-hardware/drivers/print/sample-language-monitor) topic that links from the **Language Monitors** topic, has been updated to include the location of all available versions of sample source code for the Pjlmon.dll language monitor.

closed 2020-06-17 by barrygolden