MicrosoftDocs/windows-driver-docs #2120: Visual C++ > Windows Driver > Tests cannot be found

o60816, 2020-03-26:
I would like to create the test project for the driver using c++, but according to the paragraph "To customize a driver test using the Driver Test template for C++" I cannot find the Visual C++ &gt; Windows Driver &gt; Tests in my VS 2015 even 2019, may I know where I can create the driver testing project?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6b9c14aa-06ab-bacb-7b10-ff184858d1d1
* Version Independent ID: 0cb2cc28-6e79-1732-722d-b903a82e2046
* Content: [How to write a driver test using a Driver Test template - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/how-to-write-a-driver-test-#feedback)
* Content Source: [windows-driver-docs-pr/develop/how-to-write-a-driver-test-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/how-to-write-a-driver-test-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-03-26:
Hi @o60816, I believe the WDK prior to 2015 contained this template but it never made it in to 2015 and beyond. I have no info on a replacement.  

You could try the forums --

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

Thanks for the feedback!

tedhudek, 2020-03-27:
I added a note at the top of https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/how-to-write-a-driver-test-

closed 2020-03-27 by tedhudek

o60816, 2020-03-27:
Thank you for your reply, I appreciate it very much.
May I know without this template test project how can I create my own test cases? Is it still possible to create custom test cases?

tedhudek, 2020-03-27:
Thanks for the question, @o60816.  I'm sorry to report that there is no current workaround to create custom test cases.