MicrosoftDocs/windows-driver-docs #2122: Filter by productName?

Melvincillo, 2020-03-26:
Is there any API to filter by productName or any other property besides product ID? We have more than 30,000 and we are struggling to find a product by name getting all the products and using the continuation token

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 40f09988-3ba3-0ce2-9952-b11ddd66aff1
* Version Independent ID: dd0d280c-f6a0-4a07-7f86-42eafaf6087b
* Content: [Get all products - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-all-products#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/get-all-products.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-all-products.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

tedhudek, 2020-11-16:
Hi @Melvincillo thanks for the query and apologies for the delay.

It turns out that you can do this in the dashboard web UI.  I'll have more info on the API solution for you soon.

tedhudek, 2020-11-21:
Actually there is no API available to filter products by product name.  Good luck!

closed 2020-11-21 by tedhudek