MicrosoftDocs/windows-driver-docs #2127: After copying debug version of NDIS system not booting

SJKU, 2020-03-26:
After copying the sys file to the Driver32 folder, system not able to reboot. The SrtTrail.txt shows that critical file is corrupt Ndis.sys. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0e932b09-60b4-6dd4-0b7d-9921851f049b
* Version Independent ID: eed17353-9263-be53-e497-2db6793b2da7
* Content: [Preparing for NDIS Debugging - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/preparing-for-ndis-debugging)
* Content Source: [windows-driver-docs-pr/debugger/preparing-for-ndis-debugging.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/preparing-for-ndis-debugging.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2020-03-27:
@SJKU.  Did you disabled windows file protection through registry key?
Did you booted the windows into safemode , after copying ndis.sys ?

DOMARS, 2020-06-29:
@sjku - Thanks for this issue. We have been working to remove references to checked builds, as they are not used much anymore and have not been available for Windows 10 for a long time. The previous functionality is now available in tracing as described in https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/enabling-ndis-debug-tracing. For additional information on checked builds and NDIS tracing, this OSR thread has additional information. https://community.osr.com/discussion/291983/win10-checked-version-of-ndis-sys

closed 2020-06-29 by DOMARS