MicrosoftDocs/windows-driver-docs #2128: Add link for IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG

csm10495, 2020-03-28:
I couldn't find documentation on this Ioctl.

Also can we send Identify via IOCTL_STORAGE_PROTOCOL_COMMAND. It would be nice to be able to send passthrough commands via there, like identify, that don't alter device state.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ef2ef7fe-2723-5158-f07f-992fd06516bc
* Version Independent ID: a5172bde-1554-8813-f2a3-7e0fd96370b7
* Content: [NVMe Features Supported by StorNVMe - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/stornvme-feature-support)
* Content Source: [windows-driver-docs-pr/storage/stornvme-feature-support.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/stornvme-feature-support.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

RAJU2529, 2020-03-28:
@csm10495 . Your requested link not available, but I give the link for 
**IOCTL_STORAGE_GET_DEVICE_NUMBER**
**https://docs.microsoft.com/en-us/windows/win32/api/winioctl/ni-winioctl-ioctl_storage_get_device_number?redirectedfrom=MSDN**

csm10495, 2020-03-28:
That doesn't really make sense. The doc references IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG but it isn't anywhere in the docs. Either remove the reference to it or add it to the doc.

lorihollasch, 2020-03-30:
@csm10495 , @RAJU2529 , I'll add links to appropriate IOCTL pages (and will create any missing pages) within the next few weeks!

RAJU2529, 2020-03-31:
@lorihollasch . Thanks  Author

tim-oleksii, 2020-04-23:
@lorihollasch could you please post any updates? Very much appreciated.

lorihollasch, 2020-04-23:
@tim-oleksii , @csm10495 , it turns out that IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG will ship with Windows 10, version 2004, and was mentioned prematurely on this page. V2004 will ship sometime in early May and this page will go live then.

tim-oleksii, 2020-04-23:
> @tim-oleksii , @csm10495 , it turns out that IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG will ship with Windows 10, version 2004, and was mentioned prematurely on this page. V2004 will ship sometime in early May and this page will go live then.

Thanks!

tim-oleksii, 2020-05-13:
@lorihollasch 
V2004 has been shipped, could you please describe syscall? Thanks!

lorihollasch, 2020-05-13:
@tim-oleksii , odd, the page is published, but not discoverable via internet search. Let me do some digging. In the interim, here is a link to the [IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddstor/ni-ntddstor-ioctl_storage_get_device_internal_log) page. I still need to add links from matrix pages to reference.

tim-oleksii, 2020-06-03:
Great, I see it, thanks @lorihollasch . 
There are a couple of questions still about this syscall. It it possible to provide an example? It'd be so great to see how to use it for NVMe device.

lorihollasch, 2020-08-13:
@csm10495 , thanks for opening this issue and sorry for delay in closing it. I've added links to all of the IOCTLs on this page (which I should have done in the first place).
@tim-oleksii , I added a bit of pseudocode to explain the setup - hope that helps!

closed 2020-08-13 by lorihollasch