MicrosoftDocs/windows-driver-docs #2133: tittle of this page is wrong

hotbroker, 2020-03-29:
tittle of this page is wrong

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d000568c-90a0-b9af-96c5-b33be21fb691
* Version Independent ID: aaf76b5a-7f01-8073-c07f-97c7b01a4d13
* Content: [WinDbg Preview - Time Travel - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-timetravel-preview#feedback)
* Content Source: [windows-driver-docs-pr/debugger/windbg-timetravel-preview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/windbg-timetravel-preview.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2020-03-29:
@hotbroker . Title is correct

hotbroker, 2020-03-29:
> @hotbroker . Title is correct

 i dont think so,
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-timetravel-preview

WinDbg Preview - View Menu vs WinDbg Preview – Time Travel menu

RAJU2529, 2020-03-29:
@hotbroker . if you think its correct means you can Create PR

EliotSeattle, 2020-04-10:
Thanks, @hotbroker for catching this. I've corrected the title and pushed it to the master branch. Should be live shortly.
Eliot
Cc: @DOMARS

closed 2020-04-10 by EliotSeattle