MicrosoftDocs/windows-driver-docs #2141: Methods are not working

thehustler088, 2020-04-01:
Hello, Most of the WMI Class methods are returning None(Tried with C#). It would be great if you can have example snippets

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 03977f48-1617-785e-f16a-395d23c5d677
* Version Independent ID: 51e69987-eb84-aa47-25a2-c6e496a69d10
* Content: [MS\_SMHBA\_SAS\_Port WMI Class - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/ms-smhba-sas-port-wmi-class)
* Content Source: [windows-driver-docs-pr/storage/ms-smhba-sas-port-wmi-class.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/ms-smhba-sas-port-wmi-class.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

Prabhu088, 2020-04-01:
using System;
using System.Management;
using System.Collections.Generic;

namespace test
{
    class Program
    {
        static void Main(string[] args)
        {
            String query = "select * from HBAFCPBindingEntry2";
            ManagementScope scope = new ManagementScope(@"root\wmi");
            scope.Connect();
            List<PropertyData> valueList = new List<PropertyData>();
            SelectQuery selectQuery = new SelectQuery(query);
            ManagementObjectSearcher searcher = new
            ManagementObjectSearcher(scope, selectQuery);
            ManagementObjectCollection information = searcher.Get();

            try
            {
                foreach (ManagementObject obj in information)
                {
                    foreach (PropertyData property in obj.Properties)
                    {
                        Console.WriteLine("printing object properties");
                        Console.WriteLine(property.Value.ToString());
                        }
                    }
                }
            }
          
            searcher.Dispose();
        }
    }
}

Prabhu088, 2020-05-11:
am i ever going to get support?

lorihollasch, 2020-05-11:
@Prabhu088 , what kind of support are you looking for? The original issue is a request for a code snippet, which it looks like you provided?