MicrosoftDocs/windows-driver-docs #2142: Where can I find more information about null drivers?

player55328, 2020-04-02:
Is it possible to create one that can be installed without any coding other than the INF file?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: efcada3d-1328-62f8-0374-09a0a383e1a9
* Version Independent ID: df270649-afbb-d3e5-fab2-e34351f05542
* Content: [Installing a Null Driver - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/installing-a-null-driver#feedback)
* Content Source: [windows-driver-docs-pr/install/installing-a-null-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/installing-a-null-driver.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-04-02:
Hi @player55328, I don't know much about null drivers.  Have you tried creating a driver project containing just the INF snippet on the doc page?

player55328, 2020-04-03:
I have never made a driver before. The Microsoft website makes it sound simple with just an inf file but is very vague with the details of what is required in it.

tedhudek, 2020-07-22:
Sorry for the delay. I don't have any info on this other than what's on the doc page. You could try using the snippet there and see what happens.

If that doesn't work and this is still an issue, you may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

Thanks again for reaching out!

closed 2020-07-22 by tedhudek