MicrosoftDocs/windows-driver-docs #2145: 0x91 without working Driver Verifier

MrPepka, 2020-04-04:
I noticed that the blue screen 0xC4 with the first parameter 0x91 can appear without a working Driver Verifier. The description says something like this:
The driver switched stacks using a method that is not supported by the operating system. The only supported way to extend a kernel mode stack is by using KeExpandKernelStackAndCallout.
Is the control included in the description performed by Driver Verifier or by the system? I noticed this only on Windows 7, Server 2012 and on Windows 8 and 8.1

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 48d16a94-9fc7-672c-ed1b-e0cae695d05b
* Version Independent ID: 4965db7f-5008-23f5-eba8-040f9236c4cf
* Content: [Bug Check 0xC4 DRIVER_VERIFIER_DETECTED_VIOLATION - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xc4--driver-verifier-detected-violation?redirectedfrom=MSDN)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xc4--driver-verifier-detected-violation.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xc4--driver-verifier-detected-violation.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

john-8998, 2020-05-13:
This Arg1 0x91 of DRIVER_VERIFIER_DETECTED_VIOLATION will occur (BSOD) prior to Win10, when an exception is thrown from a NMI handler. Even a try-except wont help here.
Any idea how can we catch an exception inside an NMI handler?

DOMARS, 2020-12-18:
@MrPepka and @john-8998 - Thanks for passing on what you are seeing. This particular channel is monitored by the men and women who write the driver documentation posted on docs.microsoft.com/drivers. Though we'd like to, we don’t have the resources to help you with the memory usage situation that you are encountering with driver verifier.
 
Your best bet is to use one or both of the two really good forums where other driver developers and Microsoft engineers can be found. The first is the MSDN Windows Hardware WDK and Driver Development forum: https://social.msdn.microsoft.com/Forums/windowshardware/.
The other is the NTDEV forum from OSR Online: https://community.osr.com/.
 
You can post your question in those locations, and the community of developers usually give feedback fairly quickly.
 
The memory specific examples of the debugger docs may be of some help, if they cover the usage you are encountering. For example https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/example-12---using-page-heap-verification-to-find-a-bug, describes page heap verification.

Again, thanks for reaching out. Best of luck with this issue,

closed 2020-12-18 by DOMARS