MicrosoftDocs/windows-driver-docs #2147: "SetupDiGetDeviceInterfacePropertyW function" page

Thermi, 2020-04-04:
Broken links:
* Device status and problem properties

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ec1b2f59-8582-7276-afb2-5a25d4511020
* Version Independent ID: 18cf63b3-21e3-0219-4021-7f440936d707
* Content: [System-Defined Device Properties - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/system-defined-device-properties2)
* Content Source: [windows-driver-docs-pr/install/system-defined-device-properties2.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/system-defined-device-properties2.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-05-22:
Should link to retrieving-the-status-and-problem-code-for-a-device-instance

aviviano, 2020-06-03:
Thanks @Thermi. The link is fixed.

closed 2020-06-03 by aviviano