MicrosoftDocs/windows-driver-docs #2149: PUSB_ENDPOINT_DESCRIPTOR

bobo24105, 2020-04-05:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: eb3df692-d3f4-08eb-85b9-6ac8fcdbecb1
* Version Independent ID: 2d7ba7af-3fbe-31fb-a6eb-2db80765614a
* Content: [USB endpoints and their pipes - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-endpoints-and-their-pipes)
* Content Source: [windows-driver-docs-pr/usbcon/usb-endpoints-and-their-pipes.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/usb-endpoints-and-their-pipes.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

EliotSeattle, 2020-04-09:
Hi @bobo24105 
I don't see a request or feedback about the documentation on this Issue. Was this sent by accident, or are we missing your feedback?
Thanks,
Eliot

billlattimer, 2021-01-25:
@bobo24105 closing this request as there's no actionable feedback. Please open a new issue with specific feedback if you still have questions. Thanks!

closed 2021-01-25 by billlattimer