MicrosoftDocs/windows-driver-docs #2151: Update "initial" driver package

sychevathotmail, 2020-04-06:
after the driver is published to windows update, I can't find a way to update it with newer package.  
[image] implies that there should be a way, but clicking "here" just takes me to existing driver form

EliotSeattle, 2020-04-06:
Thanks, @sychevathotmail 
Can you please tell me the page title or URL for the topic this occurs on?
Thanks,
@EliotSeattle

sychevathotmail, 2020-04-06:
Eliot, I log in with my company's credentials at https://partner.microsoft.com/en-us/dashboard/hardware/Search, then pick the previously uploaded (and now signed and published) driver package. Supposedly this page has some ways of uploading new CAB file, such that when I create new "label", I can pick new version.

sychevathotmail, 2020-04-06:
I followed guide https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/create-a-new-hardware-submission to create "initial" package

EliotSeattle, 2020-04-06:
Hi @sychevathotmail 
My team creates the topics that *describe* the dashboard. The Hardware Dev Center Support team are the right people to help investigate an issue like this. To reach out to them:
1. **First**, sign into your hardware dev center account, and then 
2. Open a Hardware Dev Center support case for investigation. 
3. Go to 
[https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq)
and look for: 
 
**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). Select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

**Please** do let me know if the documentation topics need to be updated because of your issue. 

Best of luck with this.
Eliot

closed 2020-04-06 by EliotSeattle