MicrosoftDocs/windows-driver-docs #2159: Outdated: List of Operating System Family Codes

clenzix, 2020-04-09:
Hello, I don't see Vibranium-20H1 (Vb Client) listed in the OS Family Codes list.   Can this list be updated?  Our team is attempting to create automation around the HW Dashboard API and will need latest parameters, etc.   Thank you! -Chris

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1c20769b-d969-a51a-4eb3-f9cc27875f28
* Version Independent ID: 853fe2a9-d654-98a3-7635-c30a450cdbdc
* Content: [Get product data - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-product-data#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/get-product-data.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-product-data.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-04-22:
Hi @clenzix 
Thanks for asking about this info. I've added the Operating System Family Code and the Operating System Codes. They are invalid until the operating system is released. They are:
Family Code: 
Windows_v100_VB

OS Codes:
WINDOWS_v100_VB_FULL                       Windows 10 (next update) Client x86
WINDOWS_v100_X64_VB_FULL                Windows (next update) Client x64
WINDOWS_v100_ARM64_VB_FULL          Windows (next update) Client ARM64

I'll replace "next update" with the official name, once we release that.

Thanks for opening this Issue and thanks for your patience while I pulled this info together.
Hope this helps,
Eliot

closed 2020-04-22 by EliotSeattle

clenzix, 2020-04-22:
Aeesome! Thanks @EliotSeattle