MicrosoftDocs/windows-driver-docs #2160: Broken links in HID docs

DJm00n, 2020-04-09:
I have HTTP 301 Moved Permanently code when trying to open any subpage in these pages:
https://docs.microsoft.com/windows-hardware/drivers/hid/obtaining-hid-reports
https://docs.microsoft.com/windows-hardware/drivers/hid/sending-hid-reports
https://docs.microsoft.com/windows-hardware/drivers/hid/interpreting-hid-reports
Maybe TOC is broken or something.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c16f66f8-babb-8d23-8ce5-52f9b87c26b1
* Version Independent ID: 49578792-b8e0-8a37-515e-790378ea7dfa
* Content: [Interpreting HID Reports - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/interpreting-hid-reports)
* Content Source: [windows-driver-docs-pr/hid/interpreting-hid-reports.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/hid/interpreting-hid-reports.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

DJm00n, 2020-07-01:
https://github.com/MicrosoftDocs/windows-driver-docs/commit/9f19bd5132fe04154efcb39346b864f4619e98e5
Seems these pages are removed. Why?

DJm00n, 2020-07-07:
@EliotSeattle any updates on this topic? Seems several pages get removed.

EliotSeattle, 2020-07-07:
Hi @DJm00n 
We have recovered the source for the missing pages from an earlier doc set. For some reason, these were dropped in a content migration from one source system to another. @billlattimer will research the issue and reinstate the pages, assuming they are still supported. Not sure about the schedule for this task, though, as it competes with other, hi-pri work.
Thanks,
Eliot

DJm00n, 2020-07-07:
@EliotSeattle thank you for a quick response.

Subpages of these pages was the only source on how to work with HID devices:
https://docs.microsoft.com/windows-hardware/drivers/hid/obtaining-hid-reports
https://docs.microsoft.com/windows-hardware/drivers/hid/sending-hid-reports
https://docs.microsoft.com/windows-hardware/drivers/hid/interpreting-hid-reports
Please give this task some priority.

billlattimer, 2020-08-20:
@DJm00n Thank you again for catching this drop; the pages will go live shortly!

closed 2020-08-20 by billlattimer

DJm00n, 2020-09-09:
@billlattimer hi!
These pages are still missing: https://github.com/DJm00n/windows-driver-docs/commit/a60e0bd21570a206f626c6c97c1f645bbdd89315

billlattimer, 2020-09-18:
@DJm00n this was my error - it should be live in a few hours. Thank you for your diligence to ensure this content was live!