MicrosoftDocs/windows-driver-docs #2161: SSL.com EV code signing certificate not recognized as valid for Hardware Program Registration

amagliul, 2020-04-09:
We've recently purchased an EV code signing certificate from SSL.com, as mentioned on this page:
https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-code-signing-certificate

After signing the "SignableFile.bin" file with our EV certificate and uploading, we are seeing the following error:
`"Your certificate's CA root cert is not in the approved list of root certs"`

When trying to verify the file with signtool with the `/kp` argument we see this:
& "C:\Program Files (x86)\Windows Kits\10\bin\10.0.18362.0\x64\signtool.exe" verify /kp 'D:\Downloads\SignableFile.bin'
```
File: D:\Downloads\SignableFile.bin
Index  Algorithm  Timestamp
========================================
SignTool Error: A certificate chain processed, but terminated in a root
        certificate which is not trusted by the trust provider.
```

When verifying with the `/pa` argument, the file is validated without issue:
& "C:\Program Files (x86)\Windows Kits\10\bin\10.0.18362.0\x64\signtool.exe" verify /pa 'D:\Downloads\SignableFile.bin'
```
File: D:\Downloads\SignableFile.bin
Index  Algorithm  Timestamp
========================================
0      sha256     RFC3161

Successfully verified: D:\Downloads\SignableFile.bin
```

The CA is listed on this page: https://ccadb-public.secure.force.com/microsoft/IncludedCACertificateReportForMSFT with the following information:

`SSL.com | SSL.com EV Root Certification Authority RSA R2 | 743AF0529BD032A0F44A83CDD4BAA97B7C2EC49A | 2E7BF16CC22485A7BBE2AA8696750761B0AE39BE3B2FE9D0CC6D4EF73491425C | 2017 May 31 | 2042 May 30 | RSA 4096 bits | SHA256WithRSA | Server Authentication; Client Authentication; Code Signing; Document Signing; Encrypting File System; Secure Email; Time Stamping | Included`

I have attached the signed SignableFile.bin file (in a zip), as well.

[SignableFile.zip](https://github.com/MicrosoftDocs/windows-driver-docs/files/4458910/SignableFile.zip)

I would really appreciate any insight or help that the Microsoft team can provide in this matter.

EliotSeattle, 2020-04-09:
Hi @amagliul 

This feedback form gets sent to my team, the Windows driver documentation team. We use it to track documentation issues. The Hardware Dev Center Support folks are the right people to help investigate a customer problem with submissions, like this. To open a ticket:

1. **First**, sign into your hardware dev center account, and *then*...
2. Go to [Hardware Dashboard FAQ](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq), where you'll find this passage: 
 
**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). Select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

It's important that you sign into your account first.

I hope this helps.
Eliot

closed 2020-04-09 by EliotSeattle

haephrati, 2021-11-08:
We are having the same problem with a valid EV code signing that worked until 2 months ago. Is there a new problem? We opened a support case with Microsoft but no one got back to us.