MicrosoftDocs/windows-driver-docs #2183: how to change update time?

jeffythedragonslayer, 2020-04-21:
How do I change update time? In the "Performance Monitor Properties" window, General-&gt;Graph elements has "Sample every" and "Duration:" fields, but this documentation is unclear on which field "update time" is.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 36af6504-4758-d4fd-def3-92bdecc0649d
* Version Independent ID: eb5e203b-89b6-0a76-7721-4366d978a5d9
* Content: [Determining Whether a Leak Exists - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/determining-whether-a-leak-exists#feedback)
* Content Source: [windows-driver-docs-pr/debugger/determining-whether-a-leak-exists.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/determining-whether-a-leak-exists.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

AFuckingTortoise, 2020-05-08:
Right click performance Monitor then select propertied, and change duration under graph elements.

DOMARS, 2020-05-08:
@jeffythedragonslayer - I believe the other comment on this issue describes an accurate way to update the time interval. 

The best route for questions such as this is probably in the forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk) 

Good luck with your memory leak work.

closed 2020-05-08 by DOMARS