MicrosoftDocs/windows-driver-docs #2196: "ClassVer" entry is not mentioned

mr-intj, 2020-04-23:
Some of the driver samples have an entry in their INF similar the following:
```
[Version]
ClassVer = 4.0
```

I can't find any mention of this entry or how to set it in the documentation.

It shows up in many/all printer drivers, but I've seen it in the INF files for `class=display` and `class=media` drivers as well...

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 89e65b9c-b342-6df7-e878-4b88a9590363
* Version Independent ID: 8f0d7eb7-cd13-2769-f20d-296ecc96d417
* Content: [INF Version Section - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-version-section)
* Content Source: [windows-driver-docs-pr/install/inf-version-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-version-section.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-04-23:
Good question @mr-intj.  Investigating

mr-intj, 2020-04-28:
Thank you sir. All my Google-Fu has been able to tell me is that *Printer* INFs usually specify 4.0 and *Display* and *Media* sometimes specify 2.0. I'm assuming it's referring to the class driver installer version, but I can't seem to find any more info...

tedhudek, 2020-04-29:
Added a brief blurb

closed 2020-04-29 by tedhudek