MicrosoftDocs/windows-driver-docs #2200: Not clear whether UmdfLibraryVersion implies a possibly unmet dependency

mr-intj, 2020-04-24:
In various sections of an INF file, there are various entries that provide version information, e.g. `UmdfLibraryVersion`

This begs the question "Is this a dependency that I need to worry about?". It would be helpful to add some text describing this in more detail.

For a personal example: The build on my Win10x64 dev machine populates this field with "2.25.0", but early releases of Windows 10 had versions lower than that. Is this something that I need to worry about? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6bd0d4c0-7cfd-278b-12e4-a6532269db54
* Version Independent ID: 660209eb-539f-6ad5-dc93-7879d84308a1
* Content: [Specifying WDF Directives in INF Files - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/specifying-wdf-directives-in-inf-files)
* Content Source: [windows-driver-docs-pr/wdf/specifying-wdf-directives-in-inf-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdf/specifying-wdf-directives-in-inf-files.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-04-30:
Hi @mr-intj, I think that UmdfLibraryVersion is auto-set when you choose a driver template in Visual Studio. See https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/umdf-version-history for info on UMDF versions and the corresponding Windows versions.  What OS versions are you targeting for your driver?

mr-intj, 2020-05-01:
> What OS versions are you targeting for your driver?

Right now I've only been told we're targeting "Windows 10", with no more specific criteria than that. But I'm not sure whether that's germane to my question.

> I think that UmdfLibraryVersion is auto-set when you choose a driver template in Visual Studio

Right, and once I was aware of this, the question naturally popped up "Could there be a situation where my driver is stamped with UdmfLibraryVersion N.M, but the target Win10 system only has, say, version N.M-1 on it?

If that *is* a possibility, does my installer need to deal with that (e.g. offer to install the redistributables for N.M), or does Windows do it automatically, etc.

That's the part that I didn't see covered in the docs (at least on the page I was reading)

tedhudek, 2020-05-06:
Caveat - I'm not a driver developer, so you might want to try the forums for actual dev advice. See links at bottom of this update.

Relevant doc links: 

https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/umdf-version-history
https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/building-a-wdf-driver-for-multiple-versions-of-windows

My take is that you have 2 options. You can use UMDF 2.29 (latest) with the functionality described in the second link to support users back to 2.25.  Or, you can build with the earliest Win10 version of UMDF that has the functionality you need, say 2.15, then your driver runs on Win10 v1507 and later (see first link).

Popular driver development forums:

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

closed 2020-05-06 by tedhudek