MicrosoftDocs/windows-driver-docs #2201: Ambiguous/confusing grammar syntax

mr-intj, 2020-04-24:
In the syntax definitions for INF file entries, brackets are being used to identify tokens _and_ to designate parameters as optional. 
```
reg-root, [subkey],[value-entry-name],[flags],[value][,[value]]
```

Perhaps another pair of characters can be used to enclose tokens besides brackets? Also, since leading/trailing spaces are ignored, why not use them to further improve readability?
```
reg-root, &lt;subkey&gt;, &lt;value-entry-name&gt;, &lt;flags&gt;, &lt;value&gt;[, &lt;value&gt;]
reg-root, {subkey}, {value-entry-name}, {flags}, {value}[, {value}]
```

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: acdc91cd-c89e-684c-c48b-5d66d098fa9c
* Version Independent ID: 4865d5a4-df22-1c26-f993-a661882194a7
* Content: [INF AddReg Directive - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-addreg-directive)
* Content Source: [windows-driver-docs-pr/install/inf-addreg-directive.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-addreg-directive.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-04-30:
Hi @mr-intj, square brackets always means optional in these.  The second value in your example is for multi-sz and binary ones where you may need to specify multiple values to fully describe the value.

closed 2020-04-30 by tedhudek

mr-intj, 2020-05-02:
Ah, I see. I was expecting something more like this:
```
reg-root[, (subkey | )[, (value-entry-name | )[, (flags | )[, (value | )[, (value | ) ...]]]]]
```
Thanks.