MicrosoftDocs/windows-driver-docs #2202: What about device icons?

mr-intj, 2020-04-24:
&gt; A driver package can also contain other files, such as a device installation application, _a device icon), device property pages, and so forth. For more information, see the following topics:
&gt;
&gt; * Providing Device Property Pages
&gt; * Installing a Boot-Start Driver

If you just want to provide one or more icons with your driver, neither of the two options above seems relevant.

**Context:**

I was looking for guidance on where to copy my .ico file(s). My `driver.dll` gets copied into %SystemRoot%\System32\drivers\UMDF, and on my PC that folder _only_ contains DLLs. Most other INF files I found online copy their icon files either to %SystemRoot% or %SystemRoot%\System32, but I don't like just dumping files there and hoping to avoid name collisions...

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d596ea2c-dee5-42eb-c5c3-87d74a9047b3
* Version Independent ID: 697d75cc-69f7-6790-194f-8c9478cff3c4
* Content: [Components of a Driver Package - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/components-of-a-driver-package)
* Content Source: [windows-driver-docs-pr/install/components-of-a-driver-package.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/components-of-a-driver-package.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-04-30:
Hi @mr-intj, I just updated https://docs.microsoft.com/en-us/windows-hardware/drivers/install/providing-vendor-icons-for-the-shell-and-autoplay. Let me know if it does not answer your question here and in #2204. 

Good luck!

closed 2020-04-30 by tedhudek

mr-intj, 2020-05-01:
Nice. Much better  :-)