MicrosoftDocs/windows-driver-docs #2206: Possible typo

mr-intj, 2020-04-25:
&gt; &lt;enumerator&gt;\\&lt;enumerator-specific-device-ID&gt;

Not certain, but I suspect there's only supposed to be a single backslash here(?)  Could be copypasta from source code that was escaping the backslash?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 93712a25-6177-1a6f-7e7e-cfa6100c663f
* Version Independent ID: 2fa7063a-a379-ac98-6209-5918bc257af2
* Content: [Hardware ID - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/hardware-ids)
* Content Source: [windows-driver-docs-pr/install/hardware-ids.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/hardware-ids.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-04-27:
Hi @mr-intj, thanks again for the feedback.  I'll try to address the issues you've opened as I have time. Wanted you also to be aware that you can submit pull requests if there are specific changes you'd like to suggest on pages. Doing so increases the likelihood of getting it published sooner.  To open a PR, click the **Edit** button on the docs.microsoft.com page, and then click the pencil ("edit this file") icon on GitHub.  Happy to take a look at anything you think would be worthwhile.

tedhudek, 2020-04-30:
Yes, this was a typo. Thanks for the report. Just pushed the fix, should be live soon.

closed 2020-04-30 by tedhudek