MicrosoftDocs/windows-driver-docs #2207: reporting a broken link

rakeshdn, 2020-04-25:
Link to  C++ for Kernel Mode Drivers: Pros and Cons white paper ( https://go.microsoft.com/fwlink/p/?linkid=56294) given above is broken.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9e755005-a187-149c-c4c4-423cedfe1b97
* Version Independent ID: 5db14ac5-b177-2ba7-17e4-da209902781e
* Content: [Portable - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/portable#feedback)
* Content Source: [windows-driver-docs-pr/kernel/portable.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/portable.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

EliotSeattle, 2020-04-27:
Thanks, @rakeshdn 
Searching to see where the white paper is vacationing. 
Eliot

EliotSeattle, 2020-04-27:
OK, that link is fixed now.
Thanks, @rakeshdn for pointing it out.
Eliot

closed 2020-04-27 by EliotSeattle

rakeshdn, 2020-04-27:
Thank you @EliotSeattle