MicrosoftDocs/windows-driver-docs #2210: How is this acceptable?

mr-intj, 2020-04-27:
I'm using Visual Studio 2019 and Windows Driver Kit 10.0.18346.1000.

I'm trying to write my first driver, and this is just one of many things that has me grinding my teeth and wondering what's happened to dev standards in Redmond:

```
C:\Temp&gt;devcon install ./IrvDriver.inf ROOT\\IvrVdModel
devcon failed.

C:\Temp&gt;
```

"devcon failed"? That's it? So "it works fine if you already know how to use it" is the new motto?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 483a899c-95b3-ba6b-d2cd-4cc514865852
* Version Independent ID: b0e0403e-c96a-dd69-b231-f60226ec6ecb
* Content: [Windows Device Console (Devcon.exe) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/devcon)
* Content Source: [windows-driver-docs-pr/devtest/devcon.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/devcon.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

mr-intj, 2020-04-28:
Sorry about the tone. I was having a particularly frustrating time over the last few days...

EliotSeattle, 2020-04-30:
No worries. Totally understand, @mr-intj 

I've been poking around and asking others, but as far as we can tell, your command should not fail, assuming the path locations are correct. I'll look some more to see if anyone spots some reason, though.

Agree that it would be helpful if we had better error codes to help point you in the right direction. I've passed that feedback to the engineering team that this return value is too vague to be of direct use.

The feedback mechanism in the docs goes straight to my team. We write the  driver *documentation* posted on docs.microsoft.com/drivers. Though we are adept at creating content to aid driver development, we're not *production* driver devs. I truly wish we could be of greater help in every facet of your coding experience. We will fix whatever doc issue you report, but we don’t always have the wherewithal or resources to help the many, many devs with their code. Your best bet for implementation questions is one or both of two developer forums. 

The first is the MSDN Windows Hardware WDK and Driver Development forum: [https://social.msdn.microsoft.com/Forums/windowshardware/](https://social.msdn.microsoft.com/Forums/windowshardware/).

The other is the NTDEV forum from OSR Online: [https://community.osr.com/](https://community.osr.com/).

Both of those get frequented by practicing devs, like you, who are willing to help the driver community.

All the best,
Eliot

mr-intj, 2020-04-30:
> The feedback mechanism in the docs goes straight to my team. We write the driver documentation posted on docs.microsoft.com/drivers.
> Though we are adept at creating content to aid driver development, we're not production driver devs.

Understood. My intent was to communicate my frustration with a tool in the hopes that it'd get passed along to the appropriate team there.

I've been running devcon in the debugger, stepping through the code and manually looking at the return of GetLastError to try and understand why things aren't working. When I have the bandwidth (and I understand this stuff better), I may fork a copy of devcon and bring it up to date with better logging and messaging (that code looks at least 20 years old).

-intj

tedhudek, 2020-10-23:
This page has a bunch of usage examples - https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/devcon-examples

If this is still an issue, I'd recommend asking in one of the following forums:

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

Thanks for reaching out and best of luck with this issue.

closed 2020-10-23 by tedhudek