MicrosoftDocs/windows-driver-docs #2211: Unclear whether non-user-visible token definitions must be repeated for each language

mr-intj, 2020-04-28:
&gt; If an exact match is found, Windows uses that Strings.LanguageID INF section to translate all %strkey% tokens that are defined within the INF.

&gt; The use of %strkey% tokens within INF files is not restricted to user-visible string values. These tokens can be used in any manner convenient to the INF writer, as long as each token is defined within a Strings section. For example, when you write an INF file that requires the specification of several GUIDs, it might be convenient to create a %strkey% token for each GUID, by using a meaningful name as a substitute for each such GUID value.

&gt; Specifying a set of %strkey% = "{GUID}" values in the INF file's Strings section requires you to type each explicit GUID values only once. This can help provide more readable internal INF documentation than by using explicit GUID values throughout the INF file.

So, if I do something like the following:

```
[Strings] ; Default, US English

; --- Non-Localizable ---
REG_DWORD     = 0x00010001
REG_MULTI_SZ  = 0x00010000
...

; --- Localizable ---
DEVICE_NAME     = "SpazzCo Gonkulator Adapter"
...

[Strings.080A] ; Spanish - Mexico
; Do I need to repeat the definitions of REG_* here?
```

There are at least two ways this could work:

1. The installers first choose a language section, and then find (or fail to find) any referenced tokens there. No other sections are searched when a token isn't found.
2. The installers search for each token in the most-specific language section first, and then _fall back_ to a less-specific section (e.g. [Strings.080A] and then [Strings])

If the behavior is _not_ #2, then using any non-user-visible tokens as the documentation suggests means that you have to duplicate them under every language section.

I think this should be mentioned somewhere in the documentation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5b2c9a8a-f997-dd8b-12dd-40e9dca402da
* Version Independent ID: 4e20d680-d96f-eebc-3d19-e97647e1a296
* Content: [INF Strings Section - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-strings-section)
* Content Source: [windows-driver-docs-pr/install/inf-strings-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-strings-section.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

mr-intj, 2020-04-28:
**Clarification**

Title/Subject is a bit misleading. The issue is really whether the INF interpreter in the various installers implement this "fallback" behavior described above.

An obvious use-case is with non-user-visible symbols, but visibility really has nothing to do with the question.

tedhudek, 2020-05-01:
Hi @mr-intj, currently there is no fallback, so you do need to duplicate all string tokens across all [Strings.*] sections, even those that are numeric/fixed constants that don’t actually need to be localized.

I'll add this to the doc soon.

tedhudek, 2020-05-05:
Added a note on the page, should appear on the live site momentarily.

closed 2020-05-05 by tedhudek