MicrosoftDocs/windows-driver-docs #2214: GPIO Button testing has 2 dead document links

grahamm78, 2020-04-29:
[Enter feedback here]

There are 2 document links on this page which are dead. they take you to a "Microsoft Connect Has Been Retired" page.

Power button behavior [https://aka.ms/connect-redirect?DownloadID=48735](https://aka.ms/connect-redirect?DownloadID=47452) 

GPIO buttons testing with MITT platform [https://aka.ms/connect-redirect?DownloadID=48735](https://aka.ms/connect-redirect?DownloadID=48735)

These links should be updated to the new document locations.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d0289af5-fd24-dd2e-17e2-5bb1f0ac56e7
* Version Independent ID: 23de8387-b9c2-5fcb-9f3e-17de42fb53e1
* Content: [GPIO buttons and indicators supplemental testing - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gpiobtn/gpio-buttons-and-indicators-supplemental-certification-testing-for-windows-8-1#feedback)
* Content Source: [windows-driver-docs-pr/gpiobtn/gpio-buttons-and-indicators-supplemental-certification-testing-for-windows-8-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gpiobtn/gpio-buttons-and-indicators-supplemental-certification-testing-for-windows-8-1.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

RAJU2529, 2020-05-03:
@grahamm78 . Thanks for reporting, just now I replaced two invalid links to correct links. Those are visible after approval from the author

tedhudek, 2020-10-22:
Thanks for reporting this @grahamm78 and apologies for the delay.  I was unable to locate the content that was linked to, so I have removed the links.

closed 2020-10-22 by tedhudek