MicrosoftDocs/windows-driver-docs #2221: Memory Manager Routines page is obsolete.

qsneeze, 2020-05-02:
Navigating to the page specified by the hyperlink Memory Manager Routines takes me to a page which 1) is previous version information. 2) The page only lists routines developers should not use, not all the Memory Manager routines. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 41838266-a165-d586-37d9-f42c2b9bcfca
* Version Independent ID: 047000a0-0695-27ab-c32a-05783d724530
* Content: [Windows Kernel-Mode Memory Manager - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/windows-kernel-mode-memory-manager#feedback)
* Content Source: [windows-driver-docs-pr/kernel/windows-kernel-mode-memory-manager.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/windows-kernel-mode-memory-manager.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2020-05-03:
@sysagaR . i am able to see the correct page, here is the screenshot
[image]

sysagaR, 2020-05-03:
@RAJU2529 Ok, but here's a few things to point out. Notice how the screenshot you took includes the  "Previous Windows Documentation" header.

Second, the listed routines are all marked as "system use." There is no list with all the memory management routines for drivers to use.

RAJU2529, 2020-05-03:
@sysagaR . Yes new documentation is not created for that article, once i asked to one of the MS content authors, he said Microsoft don't have policy update previous documents to new content.

Why can't you use  Memory management for windows drivers in kernel mode?
Sorry i am not a developer so I don't have knowledge about drivers .

sysagaR, 2020-05-03:
@RAJU2529 The memory management functions used by Win32 apps, like `VirtualProtect` are systems calls to the kernel. This code is not callable from kernel mode because 1) It violates basic hardware security design (calling user code from kernel mode), 2) If it was possible, it is inefficient because the call will have a lot of overhead as it calls the NT functions , ex. `NtProtectVirtualMemory` which just transitions into kernel mode anyways and calls one of the internal `MmXXX` memory management routines.

RAJU2529, 2020-05-03:
@sysagaR .  here the issues might the solution for your driver problems, so you can post your query to  https://techcommunity.microsoft.com/t5/windows-10/ct-p/Windows10.

And wait for the author of the article will reply to you .

sysagaR, 2020-05-03:
Usually @tedhudek or @EliotSeattle will update the webpage.

tedhudek, 2020-05-05:
Thanks for letting us know - page update will appear within a half hour or so

closed 2020-05-05 by tedhudek