MicrosoftDocs/windows-driver-docs #2225: Help with debugging when disk drive or related subsystem/driver is the culprit

niderost, 2020-05-04:
It seems that this specific bug check #0x154 is often related to problems with the disk drive the kernel memory store component tries to read or write memory pages from/to the page files. I.e., the kernel decides a page needs to be read from or written to disk, tries to access the disk and the disk or bus (or one of the drivers related to these hardware devices) malfunctions.
In these cases, it is often also the case that system event log entries and a memory dump file can't be written either. After all, the swap files are often stored on the same disk or via the same bus as where memory dumps and system event logs are written. In these cases, it can help to redirect the swap files and/or the system event log and memory dumps to another disk/bus, if available. Redirecting the swap file might prevent this bug check from appearing. Redirecting the system event logs and memory dumps might help get extra information on the exception when it happens.
Relocation the page file: https://winaero.com/blog/how-to-move-page-file-in-windows-10-to-another-disk/
Relocation system event log: https://serverfault.com/questions/268222/relocating-event-logs-in-windows-server-2008-r2/750423 (still works the same for Windows 10)
Relocation of the memory dump file location: https://www.technlg.net/windows/windows-crash-dump-memorydmp-file/

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 05c732ce-fd6c-98e4-19ca-f52c6dd3e7b5
* Version Independent ID: 868b8c46-2096-9d10-2f6b-f015e6a57003
* Content: [Bug Check 0x154 UNEXPECTED_STORE_EXCEPTION - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x154--unexpected-store-exception)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x154--unexpected-store-exception.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x154--unexpected-store-exception.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2020-05-05:
@niderost I saw some cases from Microsoft forum, few users are reported  NVME system driver is the culprit. to find the real culprit, take usb 64gb pen drive make it windows To Go and then boot with that pendrive.

DOMARS, 2020-07-15:
@niderost - As I mentioned in docs issue 2226, these git issues are for driver developers to provide input on the content.   Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who encounter this driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found.
• Windows Hardware WDK and Driver Development
 
This article provides more general information on configuring logs and dump files:  
https://docs.microsoft.com/en-us/windows/client-management/troubleshoot-windows-freeze

Many of the stop codes can have multiple causes. Sometimes a trend will develop with a popular OEM causing a class of failures that will be later corrected. Then other OEMs or Windows code itself will cause another set of failures. So it can be difficult to extrapolate a set of failures seen in one context, to another. We work with the OEMs to create fixes for as many as possible of these, as soon as possible. If running the filecheck and memory diagnostics does not surface any obvious cause, the OEMs source code is often required to determine and fix the issue, typically by the driver developer.

closed 2020-07-15 by DOMARS