MicrosoftDocs/windows-driver-docs #2226: Toshiba XG3 Firmware causing bug check #0x154

niderost, 2020-05-04:
I repeatedly got a bug check #0x154 on my Dell Precision 7720 with a Toshiba XG3 M.2 NVMe drive as the system drive (where page files, event logs and memory dumps are stored).
Especially, when the operating system had suspended the Toshiba disk to save power while the system kept running and later tried to reactivate the drive because of disk access, the complete system would often hang for a few seconds before becoming responsive again. A few times, the system would crash during the reactivation with bug check 0x154.
When the system crashed with bug check 0x154, addition findings were:
* No memory dump files were written to disk while dump files were configured correctly and in crashes with other bug check codes, a dump file would have been written, indicating that the 0x154 crashes were disk related.
* No events of the crash were written in the System Event Log or Application Event Log during the crash (as would normally be the case). The only events written to the System Event Log were the events related to booting the system and discovering an unexpected shutdown _after_ the crash, also indicating that the 0x154 crashes were disk related.
* When restarting after the system without powering off the laptop, the system would not boot because it didn't find the disk drive / system volume to boot from, again indicating that the 0x154 crashes were disk related.
Triggered by all these findings, I looked for known issues with my Toshiba XG3. It turned out I was running on firmware revision 57DC4102. This is the recommended/latest revision of the firmware according to the Dell Support site for my Precision 7720, but HP indicates there are known issues with this firmware: https://support.hp.com/us-en/document/c05751713 and provides a link to update the firmware. Unfortunately, I haven't been able to find any information on firmware revisions directly from Toshiba.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 05c732ce-fd6c-98e4-19ca-f52c6dd3e7b5
* Version Independent ID: 868b8c46-2096-9d10-2f6b-f015e6a57003
* Content: [Bug Check 0x154 UNEXPECTED_STORE_EXCEPTION - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x154--unexpected-store-exception)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x154--unexpected-store-exception.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x154--unexpected-store-exception.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2020-05-05:
@niderost . First the check the disk with the company's smart tool. Is this happened to a particular model by using a particular  nvme disk?
Use windows builtin chkdsk tool for C drive
use this command   CHKDSK C: /X /F /R /B 

Try to  update the latest bios firmware to laptop.  Try to use  HDD  and  recheck the problems.

malhal, 2020-05-31:
I have a THNSN5512GPU7 and it disappeared after a recent hard shutdown despite being on 57DA4105 which was supposed to fix this problem. To revive it I put it in a NVME to USB caddy and power on and off for various durations, e.g. 10mins on for 30 seconds off. This can take weeks  to work though.

DOMARS, 2020-07-15:
@niderost - Thanks for reaching out to the Windows Driver Developer Documentation team. This feedback mechanism is for requests about the driver developer documentation. Hopefully that driver developer is aware of the issue you are seeing on your laptop and working to resolve it in the source code that they ship.  Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who encounter this driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. 
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
You may also want to check with your OEM on driver updates, although it sounds like you have already done that.  I assume you have already checked for optional Windows updates. If a specific driver is indicated on the blue screen, that may provide additional information on the failure point to further investigate.

closed 2020-07-15 by DOMARS